dijia1124/PlusPlusBattery #9: Factory capacity missing

rafnov, 2025-06-13:
As in the topic. ;)
Additionally some of OnePlus rooted users report lack of possibility of setting factory capacity or recognizing it by Linux sysfs.
I'm going to further tests, but just look at Battery Guru.

dijia1124, 2025-06-14:
Just to double check, is the "factory capacity" you talk about the same as the "design capacity" at the bottom of the dashboard (root mode)?

Some devices have this under sys/class/oplus-chg/battery, while some not. Currently I'm not sure if there are other directories which can read it from. I'll try do some research.

For non-root mode, I noticed DevCheck somehow displays the design capacity but I just don't know how it works.

In BG there's a feature to manually set the design capacity. You also mentioned "setting factory capacity". I'm wondering why we would like to set it manually and then have it displayed.

You are welcome!

rafnov, 2025-06-15:
I will go for further investigation on that on free time. 
Thx for response and for your work on this! That's the only totally free app I know ATM. Put that on tests to some guys too. 😉