dijia1124/PlusPlusBattery #11: battery temperature instead of battery symbol

neokamen, 2025-07-05:
Could the temperature be put instead of the symbol? 
It would be very good. Thanks for the app, good job!

neokamen, 2025-07-05:
Sorry, I don't speak English and maybe this doesn't translate well. 
Next to the clock, I see a battery icon. Would it be possible to show the current temperature instead? 
Like replacing the icon with something like 38º, for exemple.
Thanks for replying so quickly

dijia1124, 2025-07-05:
> Sorry, I don't speak English and maybe this doesn't translate well. 
> Next to the clock, I see a battery icon. Would it be possible to show the current temperature instead? 
> Like replacing the icon with something like 38º, for exemple.
> Thanks for replying so quickly

Alright no worries I see. I will look into that cuz currently I've no idea how to implement that thing:)