dijia1124/PlusPlusBattery #14: Understanding "monitor" feature

lkblkb, 2025-07-29:
Hi, nice app, well done

I am not sure I understand the "Monitor" feature

I was expecting a history (e.g. chart) to be shown in the app, but it seems the monitor is a notification showing the stat(s) being monitored.

Is that correct?

Thanks!

- user on Pixel 6a 🔥😅

lkblkb, 2025-07-29:
Hi! Thanks for the reply

I might change "monitor" to "notification"

E.g. 
- Battery Monitor -> Battery Notification
- Monitor Entry Settings -> Notification Settings

dijia1124, 2025-07-29:
> Hi! Thanks for the reply
> 
> I might change "monitor" to "notification"
> 
> E.g. 
> - Battery Monitor -> Battery Notification
> - Monitor Entry Settings -> Notification Settings

Thanks. That's absolutely more accurate hahaha. Will include it in the next release!

dijia1124, 2025-07-29:
> Hi! Thanks for the reply
> 
> I might change "monitor" to "notification"
> 
> E.g.
> 
>     * Battery Monitor -> Battery Notification
> 
>     * Monitor Entry Settings -> Notification Settings

Hey, I'm still thinking about this. I remember there was an old app called "PerfMon", which displayed real-time system info in a floating window. A newer app called "PerfMon+" (see the attached image) has a similar use case — it shows live information in plain text, just like what PlusPlusBattery does.

I assume the "mon" in the app's name stands for "monitor", right?

But since I'm not a native English speaker, I'm not entirely sure whether "monitor" is the right word here, or if it should actually be "notification" instead.

[image]

Link: https://github.com/libxzr/PerfMon-Plus

On one hand, if I name it "notification", some users might expect it to show push alerts — like warnings when battery level drops — but that's not what this feature does.
It's a persistent notification that displays real-time battery stats, not an alert system.

On the other hand, the label in the bottom navigation bar is currently "Monitor". If I switch it to "Notification", it's a bit too long — some characters might get cut off depending on screen size or font scaling. Reducing the font too much also hurts readability...

[image]

So honestly, I’m still unsure which term is more appropriate in this case.

dijia1124, 2025-09-29:
Refactored in [v2.3.4](https://github.com/dijia1124/PlusPlusBattery/releases/tag/v2.3.4)

Now we have two options: Show in notification bar & Show in floating window, instead of just "battery monitor".

Thanks for your suggestion!