dijia1124/PlusPlusBattery #16: Feature Request: Chart for Temperature, Charge Level, and Battery Capacity

amarradi, 2025-08-08:
Hi @dijia1124 ,

Thank you for developing this great app to monitor battery status!

I’d like to suggest a new feature that could make the app even more useful:

Would it be possible to add a chart that displays the following battery values over time?

Temperature (°C)

Charge level (%)

Battery capacity (mAh)

A graph (e.g. a line chart) showing these values would help users better understand battery behavior – especially during charging, discharging, or under varying temperature conditions.

It would also be helpful to have options to select a time range (e.g. last hour, day, week) or even export the data for further analysis.

Thanks again for your great work – looking forward to hearing your thoughts!

Best regards,
amarrradi

amarradi, 2025-08-12:
> Hi Amarradi,
>
> Thanks for your suggestion. We can definitely add charts for certain types
> of data. However, I’m wondering if we should also include a background
> service to record data.
>
> The app was originally designed for OPlus devices, which are known for
> aggressively killing background tasks. That’s why I’m hesitant to implement
> a new service—it might get terminated frequently.
>
> Do you think it would still be useful if we skipped the background service
> and only displayed the charts while the app is in the foreground?

Hi,

Thanks for the feedback. Adding charts is a great idea and would definitely
enhance the user experience. Regarding background services, it's important
to note that many Android devices aggressively terminate background tasks
to extend battery life and optimize performance.

However, there are modern approaches and best practices to ensure that
background services work reliably. For example, you could use Android's
WorkManager, which efficiently and reliably manages background tasks.
WorkManager ensures that tasks are executed even if the device is restarted
or the app is closed.

Another approach could be to synchronize data at regular intervals when the
app is in the foreground. This would ensure that the data is always
up-to-date without requiring a constant background service.

Additionally, you could implement a notification system that informs users
when new data is available and encourages them to open the app and view the
charts. This way, you could continue to offer a useful feature without
compromising the app's performance on various Android devices.

What do you think of this approach? I look forward to contributing as an
idea generator and tester. However, please note that I have not worked with
Kotlin before and am only familiar with Java. Therefore, I will take it
slow at first to avoid disappointing anyone.

Best,
Amarrradi

Mein Krebstagebuch  https://www.marcusradisch.de

amarradi, 2025-08-14:
> Hi I'll try figure that out. Need some time. May I ask which device are
> you using? And do you have root access? Just for statistics cuz the app
> won't know what devices users have and whether they have root access.
> Kindly regards

Hey thanks for the message. No, I don't have root and so far I don't need
it to use the apps on the Google Pixel. As far as I know, the workmanger
doesn't need root either. According to the tutorials, it works quite well.
There's a lot to see on YouTube

https://developer.android.com/topic/libraries/architecture/workmanager?hl=de

dijia1124, 2025-08-14:
> Hey thanks for the message. No, I don't have root and so far I don't need
> it to use the apps on the Google Pixel. As far as I know, the workmanger
> doesn't need root either. According to the tutorials, it works quite well.
> There's a lot to see on YouTube
> 
> https://developer.android.com/topic/libraries/architecture/workmanager?hl=de
> 
> 
> >
> > Hi,
> >
> > Thanks for the feedback. Adding charts is a great idea and would definitely
> > enhance the user experience. Regarding background services, it's important
> > to note that many Android devices aggressively terminate background tasks
> > to extend battery life and optimize performance.
> >
> > However, there are modern approaches and best practices to ensure that
> > background services work reliably. For example, you could use Android's
> > WorkManager, which efficiently and reliably manages background tasks.
> > WorkManager ensures that tasks are executed even if the device is restarted
> > or the app is closed.
> >
> > Another approach could be to synchronize data at regular intervals when the
> > app is in the foreground. This would ensure that the data is always
> > up-to-date without requiring a constant background service.
> >
> > Additionally, you could implement a notification system that informs users
> > when new data is available and encourages them to open the app and view the
> > charts. This way, you could continue to offer a useful feature without
> > compromising the app's performance on various Android devices.
> >
> > What do you think of this approach? I look forward to contributing as an
> > idea generator and tester. However, please note that I have not worked with
> > Kotlin before and am only familiar with Java. Therefore, I will take it
> > slow at first to avoid disappointing anyone.
> >
> > Best,
> > Amarrradi
> >
> > Mein Krebstagebuch https://www.marcusradisch.de
> >
> >
> > <https://github.com/dijia1124/PlusPlusBattery/issues/16>)
> > #16 (comment)
> >
> > Hi Amarradi,
> >
> > Thanks for your suggestion. We can definitely add charts for certain types
> > of data. However, I’m wondering if we should also include a background
> > service to record data.
> >
> > The app was originally designed for OPlus devices, which are known for
> > aggressively killing background tasks. That’s why I’m hesitant to implement
> > a new service—it might get terminated frequently.
> >
> > Do you think it would still be useful if we skipped the background service
> > and only displayed the charts while the app is in the foreground?
> >
> > #16 (comment)
> > ,
> >
> > *@*.***>
> >
> > Hi I'll try figure that out. Need some time. May I ask which device are
> > you using? And do you have root access? Just for statistics cuz the app
> > won't know what devices users have and whether they have root access.
> >
> > Kindly regards

Thanks. I'll take a look. A bit busy lately.

Asking about root is just because I was wondering if there is anyone using those root-related features. Seems like most users don't have root access.

amarradi, 2025-08-14:
You could deactivate the root mode and publish it to see if anyone
complains and asks you for that